WignerQuantumOptics/Wigner_Time #2: abstract out device conversion

light-matters, 2024-10-01:
Timeline values should be specifiable in any unit and converted at the end by merging with device specifications.

light-matters, 2024-11-13:
work has started on this. `add_linear_conversion` is available (though it's in the wrong place) with tests.

light-matters, 2025-02-21:
I've rejigged the device creation/conversion (restructure branch) but I've realised that it implicitly assumes that all quantities with the same unit, have the same conversion properties. This will not scale well.

light-matters, 2025-02-21:
The above is not exactly true, but the implementation is still strange. We need some way of indicating whether a given unit is linear or something more complicated.

light-matters, 2025-02-24:
Current idea:
- 'device' tables can provide a nonlinear conversion function
- if there is no conversion function then the device is considered linear

vukics, 2025-02-26:
@light-matters 

You are right. At the moment what happens is that the full output range of the device is mapped affinely (that is, not even linearly!) to the digit space of the DAC.

That is, if we know e.g. that the MOT ”beat box” can be modulated between -200…200MHz, then digit=0 will correspond to -200, and digit=2**16-1 to +200.

This indeed tacitly assumes that the modulation voltage range of the device is -10…10V, which is not necessarily always the case.

Plus there are a few other tacit assumptions about the gain, etc.

light-matters, 2025-03-28:
Added a realistic test for taking an AOM calibration file, creating a 'device' entry, adding it to a timeline that uses AOM transparency and converting the result to 'digits'.

Should be useful!

closed 2026-08-19 by light-matters