WignerQuantumOptics/Wigner_Time #7: `np.int32` for the type of cycle count in `add_cycles`, creating a bug in the `finish` section

vukics, 2024-11-02:
**This is not reproduced on my Arch Linux box, where the type is `np.int64`. However, the problem is present on the lab machine.**

It makes that the special cycle count 2**31 overflows and wraps around, getting represented as -2147483648

Probably it has to be made explicit somewhere that we want `np.int64`.

vukics, 2024-11-05:
Apparently every conversion to `int` has the potential of downcasting `np.int64` to `np.int32`.

Such conversions happen in  `adwin.to_tuples` and `timeline.sanitize`.

vukics, 2024-11-14:
This is still a problem! Apparently, the ADwin C API (which probably underlies the Python) cannot cope with 64bit integers, either.
Should we fall back to 32bits everywhere?

light-matters, 2024-11-14:
If the C API cannot cope then I guess we have to :(. That seems like a bug though. You should file something with ADwin.

vukics, 2024-11-18:
@light-matters 2**31 actually doesn’t fit an int32, 2**31-1 being the largest positive integer. So this is still an issue in both `adwin.py` and the `*.bas` files. I suggest to simply switch to 2**31-1.

PR #18 (2024-11-18): Issue#7
(I hope)

Fixes #7
closed 2024-11-18 by light-matters