WignerQuantumOptics/Wigner_Time #8: Indexing the analog and digital switches arrays on ADwin side problematic + clarify increments of cycleCounts by 1

vukics, 2024-11-07:
analogIdx & digitalIdx can both run over the actual index range, which is an overkill

increment of endCC and startADC by 1 may not be needed?

light-matters, 2024-11-13:
Don't follow this?

vukics, 2024-11-13:
This is only on the ADwin side, specifically in `WignerTimeADwinADC.bas`, where there is a mysterious-looking `+1` in the definition of `startADC`.

Furthermore, the line
`if (cyclecount = endCC+1) then end` says ` '+1 is needed to resolve the indexing differences between ADwin and Python`, which I don’t understand (this is present in both `.bas` files).

vukics, 2026-09-16:
Addressed in 790528e — but **left open pending rig verification**, since these are ADbasic files.

## The `+1`s — already resolved, only the comment survived

Both were fixed long ago and nobody closed the loop:

- the spurious increment in `startADC` went in c9da16e, *"Removing spurious increment in WignerTimeADwinADC.bas addresses #46 and (part of) #8"*;
- `if (cyclecount = endCC+1) then end` was replaced by `if (cyclecount > endCC) then end` in 342f260.

What remained was the **commented-out line** still carrying the *"+1 is needed to resolve the indexing differences between ADwin and Python"* note — the very sentence this issue asks about. It is the last artifact of a question that no longer has a subject. Removed from both files.

## The indexing — live, and worse than "overkill"

The `until` bounds the index; the outer test did not:

```basic
if (data_10[analogIdx] = cc) then          ' <- unbounded
  do
    p2_dac(data_11[analogIdx], data_12[analogIdx], data_13[analogIdx])
    inc analogIdx
  until ( (analogIdx > analogArrayDim) or (data_10[analogIdx] > cc) )
endif
```

Once the index passes the filled region this reads past what Python wrote — and the arrays are **never cleared**, since `core.create` writes only `len(output[0])` entries. So it reads whatever a previous, longer run left there. If that stale cycle number equals `cc`, the loop fires and emits `p2_dac` with the previous run's module, channel and value.

**Why it has never bitten:** an accidental sentinel. `ADwin_Finish` rows carry cycle `2**31-1`, sort last, and match no `cc` in the event loop, so the index always has somewhere harmless to sit.

**That sentinel is not guaranteed.** Checked against real conversions:

| timeline | endCC | last analog cycle | |
|---|---|---|---|
| demo (with `finish`) | 3223626 | 2147483647 | sentinel |
| without `finish` | 3000200 | 3000200 | **none** |
| lab `prepare_sample()` | 1823826 | 2147483647 | sentinel |
| lab `prepare_sample(add_finish=False)` | 1621826 | 1621826 | **none** |

and `add_finish=False` is a supported option in the lab. The concrete failure is then: **run a timeline with `finish`, then one without, and the second run's `finish:` section replays outputs from the first.**

Fixed by bounding the outer test as the inner one already was, in both files.

I looked at doing it on the Python side instead — which would have been the better direction, per the rule that logic belongs out of the real-time program — but a terminating sentinel cannot be made clean: the cycle field is a 32-bit `long`, `2**31-1` is already taken by `ADwin_Finish`, and no value is both unmatchable during the run *and* greater than every `cc` at finish.

`sec:adwin` is updated: the listing is line-for-line identical to the `.bas` again, and the claim that the loop *"costs one comparison per channel group"* now says two. No arithmetic is added and the cost is still independent of the number of ramps in flight, so the argument that sentence supports is unaffected.

## Why this stays open

⚠️ **Not verified on hardware.** Per `KNOWN_ISSUES` §E these files have not been compiled, loaded or run here — that needs the ADbasic IDE and the machine. All the reasoning above is from the Python-side data. Two things to confirm on the rig before closing:

1. the patched programs compile and run unchanged for a normal timeline;
2. the stale-output scenario is real — run a timeline with `finish`, then one with `add_finish=False`, and check whether the old binary emits spurious outputs in the finish section.

The binary currently on the machine is built from the old source.

closed 2026-09-29 by vukics