WignerQuantumOptics/Wigner_Time #9: internal API

light-matters, 2024-11-11:
The introduction of the [polars](https://pola.rs/) library, which seems to be a general upgrade on pandas, shows how easily a 'standard' like `pandas` can change. 

I suggest that we maintain an internal API that abstracts out the current pandas dependency into some central place. This is not something that a user needs (or should) know about. They can use anything they like with our data-oriented 'objects' outside of the library. Inside the library however, we should make it easy to swap out the implementation of the features that we support.

A good test of this, would be to upgrade pandas to polars and verify that nothing changes.

vukics, 2024-11-12:
We shouldn’t go into this direction until the beta release / publication of the paper. The paper is the top priority now, that necessitates some improvement of the in-code documentation. Which in turn necessitates rationalization of the code at some points (e.g. `input.py`).

vukics, 2026-09-10:
This is strongly connected also to the pandas 3 issue, so I connected the two.

vukics, 2026-10-02:
Superseded, after the standardisation of the tracker agreed on 2026-10-02: the separation of public and internal API is now #163 (with #69, #138, #166 and #168 under it), and the dataframe backend is #167 (with #162). Its closed sub-issues (#45, #88, #119) keep their record here.

closed 2026-10-02 by vukics