WignerQuantumOptics/Wigner_Time #15: ANALOG_SUFFIXES ?

light-matters, 2024-11-14:
This variable doesn't seem necessary and probably shouldn't be hardcoded into the timeline module in any case?

If you have a timeline then you can extract the suffices from the variable names, such that they're always up to date.

vukics, 2024-11-14:
The variable is used in display_new. Plus it could be used for sanitization? (Which is the reason why I defined it in timeline.py)

light-matters, 2024-11-14:
Not sure what you mean. If these suffices are extractable from the timeline (and display uses the timeline), then why are they needed here?

More importantly, this seems like a very specific usecase. Other users would have more or fewer suffices and changing the constant in the module doesn't seem the right place. This should be at the user's layer of abstraction. In any case, if you're thinking about ADwin then it should be in that module.

closed 2025-02-19 by light-matters