WignerQuantumOptics/Wigner_Time #19: queue?

light-matters, 2024-11-15:
I'm wondering about the pros and cons of replacing `stack` with `queue` or some such?

The reasoning is that this function is used to build an experiment (i.e. at the user layer), whereas `stack` is really describing the implementation.

light-matters, 2024-11-19:
Other ideas:  `procedure`, `design` (nouns), `stage`, `build` (verbs)

light-matters, 2024-11-21:
It's been stewing in the back of my mind a bit and I like the idea of `build`.
```python
tl.build(
tl.create(...),
cl.ramp(...),
...
)
```

vukics, 2024-11-24:
I do not think that the implementation has much to do with `stack`, at least I didn't have the implementation in mind when I coined the name.

I'm not sure about `build`, in fact, `create` also "builds" in itself.

Stack has the connotation for me of something layered, which is what I meant.

vukics, 2025-03-07:
Probably we should not change the names of basic functions any more, as so much depends on them, even in notebooks.

closed 2025-03-07 by vukics