WignerQuantumOptics/Wigner_Time #22: How to handle time resolution globally?

vukics, 2024-11-18:
This affects `timeline.ramp` + `adwin.output`.

light-matters, 2024-11-18:
@vukics had the nice idea of 'filling-in' the ramp values at the ADwin stage. This way, the 'correct' time_resolution can be used.

vukics, 2024-11-19:
@light-matters How about this: we put pairs of floats into time and value to signify a ramp. We only need to be careful with the logic in the function timeline.previous.

light-matters, 2024-11-19:
That's a thought. Could be the way to go. Another option is the end point having an extra origin and function column. The main downside of using an existing column is that we can no longer coerce the types using pandas functions. I'll need to see how much that complicates the internal logic. 

In any case, we can make a custom export function as well that expands the ramp with default resolution if you want to display it /save it etc.

vukics, 2024-11-19:
So basically you want to avoid heterogeneous columns? I can accept that.

The other option is interesting. Can you have extra columns for only certain rows in a dataframe?

light-matters, 2024-11-19:
That would be my instinct, but I like the elegance of having the pairs for the ramp entries. Decisions decisions. 

I think you sort of can, but I don't think this is supported by polars and it would probably be easier to just make an extra column 'globally'. I think as long as most rows aren't changing then it's relatively cheap  to have an extra column for everything.

light-matters, 2024-11-19:
An  advantage of heterogeneous columns is that we could go back to digital 'value's just being true and false, which is kind of nice.

vukics, 2024-11-19:
And probably we could restrict inhomogeneity to the value column. For ramps, the time could be just the end time, and the value a tuple containing the start time, and the 2 boundary values. Maybe we don't even need to store the start value, because that's always the previous. This'd even have the advantage that we can put additional operations into an already existing timeline, with the consistency of ramps preserved. This wouldn't work if the ramps are spelled out when first creating the timeline.

Also, having the digital values as bools is very attractive.

light-matters, 2024-11-19:
I'll get the basic structure going for origin and then I'll have a look at this.

Restricting the complication to a single column could be persuasive.
Conceptually, it's definitely reasonable that the value column could have different types and, if so, then it would probably have to be unpacked at a later stage anyways.

I'm curious about what this would break though!

closed 2025-02-19 by light-matters