WignerQuantumOptics/Wigner_Time #28: Should we support contextless timelines?

light-matters, 2024-11-19:
Initially, 'context' was an optional column (treated differently from 'variable', 'time' and 'value') and I think there is still code which deals with the case that this column doesn't exist. Since it has become pretty fundamental, I'm thinking about phasing out this code and just assuming that these four columns are required.

Similarly, 'context' is treated differently in some code because we allow for 'None' instead of an empty string. Should we just assume that the minimum context is an empty string?

Thoguhts for later...

light-matters, 2025-03-24:
I'm thinking we should coerce all timelines to have a context. There's no real cost to this for those who don't use it and it simplifies the implementation. 

Empty string should be the minimum because this allows easy type coercion. 

Issue should persist until we make sure that 'None' is never returned for the context.

vukics, 2026-09-16:
Closed in 8c12456 — **no, and the code already did not support them.** What was missing was saying so, and the three leftovers matched the conclusion you both reached here.

### The substantive part was already done

`context` is in `timeline._SCHEMA`, so every frame the core builds carries it, and `input.__ensure_time_context` defaults it to `""`:

```
create, no context given    column=True  None present=False  values=['']
ramp onto contextless       column=True  None present=False  values=['']
the demo                    column=True  None present=False  values=['ADwin_Finish', 'MOT', ...]
```

### Three leftovers, now fixed

**A frame lacking the column raised `KeyError: 'context'`** from four frames down in `inherit.context` — no function name, no column name. `ensure_timeline` now takes `columns__required`:

```
`update` was given a frame missing the column(s) ['context']. A timeline has
['time', 'variable', 'value', 'context']; `context` is required and is the empty
string where unspecified, not absent and not `None` (#28).
```

**A null context propagated a real `None`** into every row that inherited from it — precisely the condition named here for staying open (*"until we make sure that 'None' is never returned for the context"*). Now normalised to `""`, the documented minimum, rather than carrying two spellings of "no context" through the package.

**`inherit._mask__no_context` promised a tolerance nothing honoured.** Its fallback to "every row" when the column was absent sat four lines above the `KeyError` that made that case unreachable. Removed — and since its `pd.Series` was `inherit.py`'s only use of pandas, the module's standing TODO about that dependence goes with it.

### One thing found alongside

`schema=` on `update` and `ramp` is a knob that cannot be turned. `rows_from_arguments` always yields 4-tuples, so any schema but the default raises `ValueError: N columns passed, passed data had 4 columns`, and nothing anywhere passed one. Removed from both, kept on the internal `_populate_timeline` — the same public/internal split as #45, and matching `sec:functions`, which documents neither with a `schema`.

Suite 223 → 227; demo and lab convert unchanged.

closed 2026-09-16 by vukics