WignerQuantumOptics/Wigner_Time #32: higher-level function needed in adwin.py

light-matters, 2024-11-21:
Currently, the notebooks still have to manually chain functions together to do necessary conversions.

There should be a super function that collects these together for convenience.

light-matters, 2024-11-21:
Are you already using `to_adwin` or is this the function that you're looking for?

vukics, 2024-11-24:
No, I think at the moment we use `adwin.add` and `adwin.output`.

light-matters, 2024-12-02:
@vukics If you give me a list of anything else that  you need in this function, then I can update `to_adwin` with what's missing.

vukics, 2024-12-02:
You tell me what I need :)

`to_adwin` should convert a timeline to ADwin output, that’s what I need :)

light-matters, 2024-12-02:
Lol, well fair enough.

 I do consider you the high-wizard of ADwin though, so once PR-ed, you can let me know how it fares in the real world.

light-matters, 2025-02-20:
`adwin.to_data` is currently the function for this.

closed 2025-02-20 by light-matters