WignerQuantumOptics/Wigner_Time #41: `np.int`-related issues are still present

vukics, 2024-12-05:
This line:
https://github.com/vukics/Wigner_Time/blob/bcc4d4152b2087b6d86a0341c29122d4e4187c3b/wigner_time/adwin.py#L306
results in a list of `np.int64`s on the lab2 machine (Windows 11 box).

This creates a problem in the dataframe query a few lines below.

Explicit conversion to `int` seems to solve the problem.

light-matters, 2024-12-09:
Good find, though another crazy one. Also good to know that you can explicitly link lines of code. Makes sense in hindsight.

vukics, 2025-02-20:
@light-matters this problem still persists, and this is the origin of the mysterious key-related error

light-matters, 2025-02-20:
> God created the natural numbers, the rest is the work of man

light-matters, 2025-03-27:
Is this still an issue?

vukics, 2026-09-16:
Closed in 6a27e70. To answer @light-matters' question — **it was still an issue, and the reason it looked intermittent is now clear.**

### What was happening

`to_tuples` split analogue from digital with a formatted query string:

```python
timeline.query("module in {}".format(mods_analogue))
```

`module` is an `int64` column, so `unique()` yields numpy scalars, and formatting those produces

```
module in [np.int64(3), np.int64(4)]
```

which pandas parses and then rejects with `UndefinedVariableError: name 'np' is not defined`. That is the "mysterious key-related error that seems to come from within pandas" — it comes from `pandas/core/computation/scope.py`, mentions nothing about modules, and gives no hint that the cause is a list comprehension two lines above.

**Why it was machine-dependent.** Under numpy 1.x, `str(np.int64(3))` was `"3"`, so the query worked. numpy 2.0 changed the repr. So whether it failed depended on the numpy on the box — which fits the lab2-only report exactly.

### What was already there, and why it wasn't enough

The `int(x)` fix from this issue *was* applied, and it is load-bearing on every export — with real demo data the column is `int64`, so removing it reproduces the failure immediately. But it was an unexplained one-word guard on a line whose fragility is invisible, and it covered only `mods_analogue`; `mods_digital` escaped by luck, because `modules__digital` builds its list with `enumerate`.

### The fix

Replaced with `wt_frame.subframe`, which filters on `isin`. Comparing values cannot have this failure mode at all, so the bug class goes rather than being guarded — and it routes the operation through `wt_frame` as the conventions require, where `.query` did not. It is also the helper `adwin/display.py` already uses for the same job.

### Regression test

Added, and **verified to be meaningful**: restoring the `.query` form makes it fail with the same `UndefinedVariableError`. It asserts the premise (the column really is `int64`) and then the separation itself — digital tuples on module 1, analogue on the rest, no row dropped.

Separately checked: no removed numpy aliases (`np.int`, `np.float`, `np.bool`, `np.object`, `np.str`) exist anywhere in `src/` or `test/`.

Suite 227 → 228; demo and lab convert unchanged.

closed 2026-09-16 by vukics