WignerQuantumOptics/Wigner_Time #42: Conversion to `np.array` seems to work differently on Windows boxes than on Linux

vukics, 2024-12-05:
So this line:
https://github.com/vukics/Wigner_Time/blob/bcc4d4152b2087b6d86a0341c29122d4e4187c3b/wigner_time/adwin.py#L133
leads to a
`ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimension. The detected shapa was (2,) + inhomogeneous part.`
This occurs on both lab machines (Windows 10 & 11)
Workaround:
`cycles = np.array([i[0] for i in output[0]] + [i[0] for i in output[1]])`

light-matters, 2024-12-09:
If this is OS dependent then that's utterly mad!

vukics, 2024-12-11:
@light-matters I have now tried this line with an output generated on my Arch Linux box, and it raises the same error.

I just assumed that you had tried that on Linux and it worked fine, that’s why I thought the error on Windows means that it works differently there.

light-matters, 2024-12-12:
(note from oral conversation)

This was tested using different origin data and worked on my machine. The particular data seems to be the problem.

closed 2025-02-20 by light-matters