WignerQuantumOptics/Wigner_Time #53: do we want 'diagnostics.py' in the package?

light-matters, 2025-02-20:


vukics, 2025-02-20:
Probably not, but how about `experiment.py`?
The latter could be kept as demonstration?

light-matters, 2025-02-20:
Yes, we should definitely keep something for demonstration. I would rename `experiment.py` and include a demo of the usage (i.e. an example of  creating a full timeline using the provided functions).

vukics, 2025-02-20:
The absorption imaging in `diagnostics.py` is a good example of something different from simple stacking of experimental stages, since we want to be able to anchor it to any context within an already existing timeline.

So, perhaps you could keep `trigger_camera`, moving it to `experiment.py` (that you may rename) which uses this feature, e.g. to record fluorescence at a given point of the timeline.

vukics, 2025-02-24:
@light-matters  What is the present status of this? At the moment I see all the three of `experiment.py`, `diagnostics.py`, and `demostration.py` in the upstream.

light-matters, 2025-02-24:
Not implemented yet. This issue will be closed on completion.

vukics, 2025-03-16:
I’ve consolidated a bit the `doc` folder, since I realized that we still hadn’t had sensible and up-to-date demo for the package.
This is in commits cc898d9 e0fe98a and 8825286

light-matters, 2025-03-24:
@vukics Please **run tests** before pushing to the main branch (especially without review)!

If files are renamed then update all of the references. 

After publication we should also disable direct pushes to main.

light-matters, 2025-03-27:
Shall be implemented in the `restructure` branch

vukics, 2026-09-23:
Settled and done in 474b7f9, on branch `claude_code`. The question was never whether to ship the lab's `diagnostics.py` as a second demo. It was whether the demo shows **interweaving** at all, and it didn't: every stage in `demo/full_experiment.py` follows the one before it, and the only `origin` given was `MOT`'s absolute `0.0`.

**What was added.** `trigger_camera`, with exactly the signature shown in `sec:stacking`, plus a camera trigger connection (`["trigger_camera", 1, 0]`, as in the lab). `test_demo.py` builds the complete demo timeline, `finish` included, and then places the trigger into it with `origin="molasses"`. That is the pattern the lab uses in `diagnosticsStageByStage.ipynb` for imaging after each preparation stage. The test checks three things: the trigger lands 2 ms after the molasses anchor, inside magnetic trapping and before the final ramps; every existing row is unchanged; and the new rows are in their own context.

Two details worth knowing:
- The camera is **not** added to `default_state`, as `sec:forwarding` argues. It is set through `init(trigger_camera=0)` / `finish(trigger_camera=0)`. `timeline__demo` is unchanged.
- The connection matters. Without it, export drops both trigger rows without any warning.

**Two paper findings from this, recorded in `KNOWN_ISSUES.md` and not yet changed in `main.tex`:**
- `sec:stacking` calls `trigger_camera(0.0,1e-3)`, but that function's `context` has no default, so as printed it raises `TypeError`. The required `context` is correct: a placed stage that inherited its context would take `ADwin_Finish` when attached to a finished timeline. So the call should change, not the signature.
- `sec:interweaving` sends readers to `sec:demonstration` for further examples, but that listing contains none.

closed 2026-09-23 by vukics