WignerQuantumOptics/Wigner_Time #56: Interwoven ramps do not find origin value correctly.

vukics, 2025-03-19:
Minimal working example:
```
timeline=tl.stack(
    ex.init(),
    ex.MOT(duration=1),
    ex.MOT_detunedGrowth(),
    tl.ramp(t=1,duration=0.1,lockbox_MOT__MHz=-2),
    tl.ramp(t=0.5,duration=0.1,lockbox_MOT__MHz=-1)
)
```
creates non-continuous curve for the `lockbox_MOT__MHz`. However, here at least some initial value is found (the -2).

In the example we saw in the lab, no initial value was found.

vukics, 2025-03-19:
I think the behavior we want in this case is continuity at the beginning of the new, interwoven ramp, whereas the user should be responsible for continuity at the end, by inserting another ramp at need at a later time.

vukics, 2025-03-19:
So basically the system should look not at the value at the altogether latest time instant, but the value at the *latest time instant before the start* of the interwoven ramp.

light-matters, 2025-03-26:
This could be working in the 'random-ramp' branch. Have a look at test_random_ramp.

(fcaab51c863a5c61ed44b9f4d0407af91048d870)

light-matters, 2025-03-26:
There may still be some debatable corner cases. Currently, if adding data to a timeline involves setting an origin, values after the origin are filtered out of the history (implementation reasons). But, maybe values after origin + input value should be filtered out. This is tricky to fix because time origin and value origins are returned in the same section of code. To change this, we would have to update all of  the time entries and then go back and check what was the 'last' value.

vukics, 2025-03-26:
fcaab51 seems to be OK with interwoven ramps, but with
0133c79 my minimal working example at the top throws
`ValueError: Ramp value less than 1e-15 detected. We assume that you don't want this!`

light-matters, 2025-03-27:
thanks, missing `np.abs`... relating to another issue

vukics, 2025-03-27:
I’m still experiencing issues, e.g. with this timeline:
``
timeline=tl.stack(
    ex.init(),
    ex.MOT(duration=1),
    ex.MOT_detunedGrowth(),
    tl.ramp(t=1,duration=0.1,lockbox_MOT__MHz=-2),
    tl.ramp(t=0.5,duration=0.1,lockbox_MOT__MHz=-1),
    tl.ramp(t=0.75,duration=0.1,lockbox_MOT__MHz=-5),
)

where I’d like to ramp back to make the variable continuous.

light-matters, 2025-03-27:
The error message is correct this time, but the uderlying logic is wrong. I think we're hitting the issue that I mentioned above. I think the anchor is being detected (as the origin), and then everything after this time is being ignored when it comes to finding the previous value. This means that the ramp is moving between two values that are the same and so you get the error. 

Clearly, this won't do and I'll have to find a better way of filtering :(.

light-matters, 2025-03-27:
@vukics Should work now?

light-matters, 2025-03-27:
works now?

light-matters, 2025-03-27:
Closed in https://github.com/WignerQuantumOptics/Wigner_Time/pull/61

vukics, 2025-04-05:
@light-matters The merge of the random-ramp branch (ee5cc0f4) seems to have introduced some new bug. With `prepareSample`, the ramps for the magnetic trapping and the final ramps dissapeared.
This can be diagnosed with the line
```
dp.display(tl.expand(ex.prepareSample(MOT_duration=1),time_resolution=25e-6))
```
in `demo.ipynb`. Cf. the attached before/after plots.

[image]

[image]

light-matters, 2025-04-14:
Context  inheritance based on `time` seems to be a problem ...

Options:
- inherit based on previous row (explicitly required otherwise)
- don't inherit at all (use argument to stack for convenience?)
- decide with a global flag?

vukics, 2025-04-14:
Option 1 is what we talked about, but I like option 2 as well.
I’m not sure about option 3, how global do you mean?

light-matters, 2025-04-15:
i was thinking at package level, but let's forget about option 3 (life's too short etc!).

Playing with option 2 has led to some interesting discoveries. You can do more with lambdas than I thought. It looks like it's fairly straightforward to create lambdas, such that the function kwargs are updated later on, e.g. in `stack`. Already useful, but it might help with some inconveniences with respect to other things you mentioned a while ago regarding ramp functions and  `expand` (vague memory).

light-matters, 2025-04-15:
e.g. we can do this:
```python
    if timeline is None:
        kwargs = dict(
            duration=duration,
            t=t,
            t2=t2,
            context=context,
            origin=origin,
            origin2=origin2,
            schema=schema,
            function=function,
            **vtvc_dict,
 )
        return lambda x,**kwargs__new: ramp(
            timeline=x,
            **{**kwargs, **kwargs__new}
        )

`

PR #72 (2025-07-08): Context inheritance
Fixes the main recent bug (disappearing ramps) as well as minor bugs regarding context inheritance.

In the end, this PR reverts the default behaviour of context inheritance, such that the last context of the previous timeline is inherited but with extra features. Particularly, `stack` has been upgraded, such that you can pass kwargs to all subsidiary functions, like the `context`. This required a fairly intensive upgrade to all of the lambdifiable functions within `timeline`. In this spirit, `expand` was also upgraded, so that `expand` can be used within a `stack`, specify [... cut]