WignerQuantumOptics/Wigner_Time #59: Display - don't plot lines by default (bad habit of scientists)

light-matters, 2025-03-25:
At least make the plotting of lines an easy flag to change.

Where there are a lot of points, it's not necessary and in a lot of cases, plottitng lines gives a false sense of what's happening through time.

vukics, 2025-03-25:
I agree that it can be made optional, but I’m not sure I get why we wouldn’t want lines.

Since there are markers in addition to lines, it is easy to spot where the actual switches happen, but lines do convey what the value is between switches.

light-matters, 2025-03-26:
The digital plots have carefully chosen 'lines' to be honest to the data. Lineplots in general however, 'fill in the blanks' between data points and so are not a reflection of the underlying data. They force a relationship that might not be there. In general thereore, I think it's bad practice to draw straight lines between data points - unless you're just checking something quickly or are doing it because there are too many data points (a kind of average).

I'm not sure what you mean by convey what the value is between switches because I would say that this is exactly what they don't do. This issue came up because I was testing ramps and had to realize that the triangles between data points were often false artifacts.

Optional is probably the best compromise.

vukics, 2025-03-26:
This is not (measured) data that we plot here.

We know that the values of variables remain constant in between the actual switches (represented by markers), and the lines should represent this, so that when I zoom in to any part of the plot, I can read the value of *all the variables*, even those that are not updated during that time interval.

I agree that the triangles in the plots of analog variables are problematic, but this is not a problem of the lines, but the way we plot – before plotting, we should fill in the blanks somehow, so that triangles don’t occur. Probably a `pyplot.step` plot will do, just like for the digital variables?

In fact, the step-plot probably represents most faithfully what ADwin actually does (keep the values constant between switches), so we should use that also for analog variables.

light-matters, 2025-03-26:
I should have made it clear initially that I was only talking about the analogue plots, apologies.

Being able to read the values of all variables while zooming is a good argument. A step function sounds like the best option by default. We just have to check that this doesn't cause performance issues - don't think it will though.

closed 2026-08-19 by light-matters