WignerQuantumOptics/Wigner_Time #65: Do we need a default time resolution in ramp functions?

vukics, 2025-03-28:
https://github.com/WignerQuantumOptics/Wigner_Time/blob/ee5cc0f4429c8d0379f584b88785c8aa3fb22857/wigner_time/ramp_function.py#L10C30-L10C61

I feel like this goes a bit against the principle of compressed ramps.

light-matters, 2025-03-31:
'Tis a fair question.

Forcing the user to always use compressed ramps might be a hindrance.

 It is already slightly different in the `restructure` branch: with the resolution default being global to the package. Therefore all time resolutions can be changed at once, when known.

 Might be worth thinking about this though.

vukics, 2026-09-29:
**The first half is done** in d9d8453 on `issue#85` (P3 of #85). A ramp now keeps a resolution it binds:

```python
tl.ramp(coil__A=1.0, duration=1e-3,
        function=functools.partial(ramp_function.tanh, time_resolution=1e-4))
```

This ramp is sampled every 100 µs by `expand` and by `convert`, whatever the cycle period. `expand`'s `time_resolution`, and `convert`'s, are now *defaults* for the ramps that bind none, never overrides (settled by @vukics on 2026-09-29). Before this, a `partial` still declared the keyword, so it was handed the cycle period silently. The ramp functions' own default, `config.TIME_RESOLUTION`, is gone.

**The second half is open:** staggering the starts of long ramps that run at the same time, so that their updates do not all fall into the same cycles.

---
*Drafted by Claude Code at @vukics's request.*

vukics, 2026-10-02:
Retitled to the half that remains. The first half – a ramp keeps the time resolution it binds – is done in d9d8453 (see the comment above), and with #85 closed this is no longer its sub-issue.

light-matters, 2026-10-02:
Is it possible to predict this? I.e. can we estimate based on ADwin's memory etc how many points is too many and ask the user to confirm a stagger within a certain resolution before upload?

vukics, 2026-10-02:
My current understanding is that the limiting factor is memory access: large arrays (and, I think the typical timeline qualifies as “large”) live in ADwin’s “slow” memory.
We can explore whether we can do anything about this, e.g. with some kind of compressed storage within ADwin’s long int type.