WignerQuantumOptics/Wigner_Time #67: Create a `pipeline` structure

light-matters, 2025-03-31:
I refactored experimentDemo (See 'demonstration.py') to make a structure for yielding different stages of an experiment on demand. This should be more extensible than the stage system.

Maybe this should be made into an official module.

light-matters, 2025-03-31:
In hindsight, it might be simpler to just return a section of a timeline with custom initialization/finalization?

vukics, 2025-04-01:
Great, I’ve hardly ever used `yield`, so I’m intrigued, and you have to explain to me the pipeline structure.

light-matters, 2025-04-01:
> Great, I’ve hardly ever used `yield`, so I’m intrigued, and you have to explain to me the pipeline structure.

You can have a look at projects/Wigner_Time/doc/demo__full_experiment.py to see it in action at the moment (restructure branch).

I'm having second thoughts though. What is the advantage of doing this over just returning the section of the 'full' timeline that we need, i.e. up to the specified context (plus finalization) ?

vukics, 2025-04-07:
Now that I've looked into `yield` a bit, I also think that this is not really a use case for that.

However, I think generators defined with `yield` will be very useful when creating a time-of-flight sequence, where the absorption imaging is stacked on the same timeline with different delays that we iterate over a range.

closed 2026-08-19 by light-matters